m5stack/M5Stack #216: (!IMPORTANT! but not this repository)SSL certificate error on M5 forum

sysdl132, 2020-04-17:
[image]
image1

The forum has been blocked on msedge(image1) 

[maybe you can reopen it on discourse(recommended)](https://discourse.org)

sysdl132, 2020-04-17:
If I click on "continue",(please see the image below this line)
[image]

sysdl132, 2020-04-17:
Data of my browser:
msedge on chromium(80)
chrome(80)
firefox(74)
nwjs(chromium 79)
(4 more)
OS:windows 10

EeeeBin, 2020-04-17:
Thanks for your reminder，we will find a way to solve this problem

sysdl132, 2020-04-30:
I think the forum is can be normally visited now.Thanks for your support.

sysdl132, 2020-04-30:
You can also open your sub forum at discourse.
[discourse website](https://discourse.org)
Your nodebb forum here:
[image]
discourse meta:
[image]

closed 2020-04-30 by sysdl132

sysdl132, 2020-04-30:
https://meta.discourse.org/t/how-to-set-up-a-discourse-forum-for-production-use-yunohost/148222?u=sysdl132