m5stack/M5Stack #222: Lora 868 module with AI Thinker RA-01H Lora chip has Reset hard wired to 26 and DI0 wired to 36

ifrew, 2020-06-23:
Just received three Lora 868 modules from you today. I discovered after 4 hours trying to get these to work with my existing code is that the LORA pin definitions in the provided Lora.h file in the M5Stack Library has the wrong pin definitions for this module. I traced the wiring on the pcb and Reset for the RA-01H goes to 26 and dio0 to 36. I saw a change made two years ago where Jimmy  reversed the pin definitions. This I believe was for a different module possible. Either way, simply changing these definitions got my code to work. My code uses interrupt handing so I need DI0 wired up correctly. The example code to read data uses a polling mechanism to read the interrupt registers which is why it works. There is no working example provided for receiver callback handing. Easy mistake but it should be corrected.

Cheers

Iain

Gitshaoxiang, 2020-06-28:
thank you feedback. we will fix the lib pin define mistake

Zontex, 2020-11-12:
> interrupt

Hi @ifrew, in the LoRa.h there is a function called setPins() which can be used to modify the pins, if you need to use them in different way you can just modify using the following command:
```
// override the default CS, reset, and IRQ pins (optional)
LoRa.setPins(ss,reset,dio0);// set CS, reset, IRQ pin
```
There is no need to modify LoRa.h file source code, the driver is correctly designed by the example given in the M5Stack library, hope it helps.

closed 2020-11-12 by Zontex