m5stack/M5Stack #226: M5Stack Grey wake on motion / movement with MPU6886 from deepsleep

geronimoooooooo, 2020-08-20:
Hey,

Few weeks ago I saw a great example that could wake up a M5StickC from deep sleep when motion was detected. You can see the video here: https://www.youtube.com/watch?v=v5GpsvjFsEw
first code version: https://gist.github.com/standarddeviant/ea0b7f12a32bf5de96992a8ef350351d
second code version: https://gist.github.com/standarddeviant/85c31cf34eb51e10aa3bb02dcd0bcbd1
integrated in M5StickC lib: https://github.com/m5stack/M5StickC/blob/master/examples/Advanced/IMU_Wake_On_Motion/IMU_Wake_On_Motion.ino

That is possible because the interrupt of the MPU6886 (SYS_INT) is wired to the GPIO35 of the M5StickC.
You can see the schematics here: https://docs.m5stack.com/#/en/core/m5stickc
You can see in the right corner of the schematic that mpu6886, rtc, esp32(GPIO35) all have SYS_INT.

Since the M5Stack Grey also uses a MPU6886, I have been trying (by modifying the M5StickC example code) now for some time to get the same functionality on my M5Stack Grey. But it doesn't work... All I want is to wake up the device from deep sleep when motion is detected. Like in the shown video with the M5StickC.

Is this even possible with the M5Stack Grey? Is there any wiring (like in the StickC) that could allow the mpu6886 to wake up the M5Stack GREY from deep sleep?

// enable waking up on pin 35 (from IMU)
esp_sleep_enable_ext0_wakeup(GPIO_NUM_35, LOW); //1 = High, 0 = Low

When I read the GPIO35 on the M5StickC, when there is no motion, it reads 1 (HIGH). As soon as I move it, it starts reading 0. As it should.
But when I read the GPIO35 on the M5Stack Grey, it reads all the time 0.

Any help would be greatly appreciated!

EeeeBin, 2020-08-20:
HI, geronimoooooooo
Unfortunately, M5Stack Grey cannot be awakened by mpu6886 
mpu6886 only used I2C pin, not connect grey with GPIO35
[image]

felmue, 2020-08-20:
Hi @geronimoooooooo 

as @EeeeBin already stated in the M5Stacks the interrupt line from MPU6886 is not connected at all, unfortunately. But it can be fixed by soldering a wire:

https://community.m5stack.com/topic/2039/wake-up-on-pick-up/15

Felix

geronimoooooooo, 2020-08-26:
Hey

@EeeeBin big thx for the very fast answer!!

@felmue awesome work! Do you have maybe a m5stack grey that you could also open and solder/wire? maybe make some images too :D

In the next version of this product there needs to be a wiring between the IMU interrupt and a pin!

felmue, 2020-08-26:
Hi @geronimoooooooo 
thanks. The same daughter board is used on M5Stack Fire and M5Stack Gray so the picture I've made for the Fire also applies for the Gray.
Cheers
Felix
P.S. I've added that information (and more) to my collection about M5Stack as well: https://www.gwendesign.ch/kb/m5stack/m5stack/#wake-on-motion-wom-using-mpu6886-interrupt

Zontex, 2020-11-12:
Hi Guys, look like answers were given, MPU not connect by GPIO only I2C, can be manually changed by soldering at your own risk.

closed 2020-11-12 by Zontex