m5stack/M5Stack #229: M5Stack grey core consumes 0.93 mA when powered off!

ifrew, 2020-09-03:
Hi ,

I've now started building my design for people and some are reporting that the battery is flat after a few days of not using it. 
It seems to vary in time but tonight I switched mine off using the 2 click method and then measured if any current was being drawn by the module. I measured 0.92mA on two different devices by grounding the wire from my proto board to the module which is connected to a battery and then using a multimeter from the battery pin on the proto board to the battery pin on the m5stack module? I wasnt expecting to see any current draw. I have read about varying issues with the Ip5306 power management chip.

I just started using the power library since I'm using a  battery. I can't remember seeing this issue before and wonder if using this library changes the internal state of the ip5306 chip such that it is always drawing some current?

Has anybody else seen this or have a work around? I'm thinking of replacing the push button power switch to a mechanical switch that is connected directly to the battery feed.

Thanks for any help you can provide

cheers

Iain

felmue, 2020-09-14:
Hi Iain 

I did some measurements with the M5Stack Gray (light sleep, deep sleep and powered off) and found the same current (about 930 uA) while powered off. Measuring IP5306 output (pin 8) while powered off interestingly reads about 1.4 volts. It almost looks like IP5306 doesn't fully shutdown for some reason. So far I have not been able to figure out why that is though.

You've mentioned that you did not notice that before starting to use the power library. Maybe you could go back to your old code base and redo the measurement?

You can find my measurements (and more) here: https://www.gwendesign.com/kb/m5stack/m5stack/#sleep-current

Thanks
Felix

ifrew, 2020-09-14:
Thanks for confirming Felix. I power my device with a 18650 Panasonic 3200mAh battery. I am finding my device goes flaf pretty quickly but not on all devices which is strange. I had heard issues with the ip5306 chip and they have now moved to using axp192 in the new core 2. However, I need the hard buttons ND no touch screen and the inbuilt imu to the core itself rather than via the connector. I was going to add a physical switch between the battery and the batt connector but then I'd have to remember to witch it on to charge while using a USB cable which for end user experience of my design isn't useful.

I am surprised though that if it's only drawing 1ma when off that it goes flat in a day or so? If it's outputting 1.4v that seems high? I'll try commenting out the power module calls and see if the default state of the ip5306 changes.

Cheers
Iain

felmue, 2020-09-15:
Hi Iain

I ram some more experiments and found the following: to get rid of the 930 uA the following bit `BOOST_ON_LOAD_BIT` needs to be cleared. That can be done by calling `setAutoBootOnLoad(false);` in function `POWER::powerOFF()`.

Unfortunately there is a side effect as the M5Stack can no longer be turned on via power button. The only way to turn it back on is either dis- and reconnect the battery or plugging in USB. Not really ideal. It looks like IP5306, with that bit cleared, turns itself completely off and no longer is monitoring the power button.

With a 3200 mAh battery your devices should last much longer. Are you sure the batteries are ok and charged? Or maybe M5Stack sometimes turns itself on accidentally?

Cheers
Felix

ifrew, 2020-09-15:
Thanks for that!. Well I found an issue. I use the m5stack lora 868 module in conjunction with my proto board. The lora 868 board sits on top of my proto board which is then mounted in my case where the battery is and the core sits on top. When I measured the voltage on the Lora 868 pins I was only getting ½ the required voltage voltage! Removing the Lora 868 module and measuring the voltage on my proto board pins I got the full voltage again! I replaced the Lora module with another I have and the voltage was back as expected. So the battery drain was due to a faulty Lora 868 board. I’ll have a look to see what the cause of that was. Measuring the current draw its backup at 950 uA. Hiwever it seems to cycle between that peak and a lower couple of values around 750 and 625.

Either way, glad I found the case of my issue. I sill close this now.

Cheers
Iain

ifrew, 2020-09-21:
Hi Felix,

Some more feedback for you and I think I have a solution now that may work for me. After researching all the threds on google I could find and seeing your cooment wrt setting autobootonload false, The following code now still only puuls about 1mA of current but the output on pin 8 is now only 0.06V and I can start it on battery using the power switch!!! This is on  core grey module with build date of 2018.3. Now seeing how long this will last on battery. 

        m5.Power.begin();
	m5.Power.setPowerBoostKeepOn(false);
	m5.Power.setLowPowerShutdownTime(m5.Power.SHUTDOWN_8S);
	m5.Power.setAutoBootOnLoad(false);
	m5.Power.setPowerBtnEn(true);

felmue, 2020-09-22:
Hi Iain

thank you for the feedback. I hope the battery life is now close to what can be expected with your solution.

I think I found the reason why the power button might not work if the unit has turned itself completely off. In the reference design of IP5306 the power button is connected with a series resistor to GND. I.e. when not pressed the input is floating and when pressed pulled down to GND through the resistor.

However in the M5Stack the power button also doubles as reset button for the ESP32 and for that to work it has been wired differently and a pull-up resistor has been added. Now if there is at least some voltage left, pressing the power button still can change the potential on the wire a little bit and the IP5306 turns on again. But if there is no voltage left, pressing the power button doesn't change the potential enough (or at all) and the IP5306 output stays off.

BTW: Maybe this is the reason why in M5StickC and the new M5Core2 there are two buttons, one for power and one for reset.

Cheers
Felix

ifrew, 2020-09-28:
Well Félix, not having much luck here. The battery is still getting drained. I now have a little test setup of only using the core attached to a bare proto board that only has the battery attached and I'm measuring the loss. I expect it will only last three days on a good Panasonic 3200 mah battery which is really strange since the current draw I measure is around 1mA and Vout from the ip5306 chip is only 0.06v. I have no idea how people are getting long battery life putting the esp32 to sleep as even with these measurements it should last longer than 3days.  I'm now thinking of using a 3 pole switch to turn on and off the device completely. However when off I still want to be able to charge the battery. So I'm thinking when the switch is in the off position if I hook up a diode to the battery connector mbus I should be able to allow it to be charged when a USB cable is attached and when not the battery won't be drained. Not sure of what type of diode to use here. Any thoughts?

felmue, 2020-09-29:
Hi Iain

something seems to be way off in your setup. Given the numbers, a 3200 mAh battery and 1 mA power consumption should give you much more than three days. Simple math would suggest something in the area of 3200 h (about 130 days)

In comparison I have an application running on a M5Stack Gray (on a 700 mAh battery) which goes into light sleep (about 7 mA) when not used and the battery lasts for about 100h (about 4 days), which matches the math quite nicely.

Are you sure your battery in good condition and is fully charged at the beginning of your experiment? Did you measure the voltage of the battery when the M5Stack Gray stopped working? Is the battery really empty?

Thanks
Felix

ifrew, 2020-10-09:
Hi Felix,

 I got sidetracked trying to get a 3 pole switch driving the 5v line to work with the m5stack using two diodes, one to ensure no battery drain when off but allowed charging and another to avoid 5v mbus line going direct to battery when usb attached. unfortunately this approach wont turn the unit on, so back to square 1. I fully charged a fresh panasonic ncr18650BE battery, 3400mAh, 3.7 V , put it in a battery holder and soldered to the battery and ground pins of a bare proto board. The module I am using is m5stack gray build date 2018.3. Here are the results so far from last 2 days attached as a picture. Just got 8 new m5stacks which arrived and build date of 1220 v2.4. Looking at the components through a magnifying glass to see if I could spot any changes, I did see that a 90K resistor in the 2018.3 pcb that i measured the resistance of next to the power switch has been removed. I wonder if this was the pullup resistor that you mentioned before? I will try the same test with this new module tomorrow after I recharge the battery to see if there is any difference. Something wrong somewhere. The batteries do not loose charge when fullyy charge up and not attached. I also checked for that. I am measuring a resistance of 0.5ohms from the mbus bat pin to the bat pin of the ip5306 which I didn't expect to see as i dont see that in the schematics anywhere. Although, in one of the data sheets for the ip5306 I did see such a resistor being used. So not sure what is going on there.
[image]

ifrew, 2020-10-09:
As a quick update, I though Id test the new m5 modules I just received. Using the battery that came with it I charged it up and measure the voltage on the 5v line..0v with about 0.005v on the 3.3v line. Totally different from the module I am testing with. I then used the module I am testing with the battery baseplate and it was showing the same as the picture above, 0.55 volts on the 5v line. I then flashed the new module I received with my software and measured the voltage again. Still 0! Finally., finally conclusive proof that some of the m5stacks are slowly draining the battery I expect due to a bad ip5306 chip or the buck charger leaking current into the m5stack and components using that current like the speaker amp. I have a few 3.2018 modules here so I can now test them to see if they are all doing the same. I should have checked that much much sooner. Ah well, you live and learn. I will check all the new modules using the voltage levels on the 5v rail to see if any of them are showing a voltage on the 5v rail. Feeling so much better now that I have a test,

felmue, 2020-10-10:
Hi Iain

thank you very much for the information - it's interesting that M5Stack seems to have fixed that issue with a newer hardware revision. And yes, the pullup resistor I was talking about is 100k according to the schematics. The available documents about M5Stack Gray do not (yet) list another hardware revision. Also, I am curious about M5Stack support's response to your inquiry. Hopefully they can shed some light into this mystery.

Cheers
Felix