m5stack/M5Stack #231: ToF_VL53L0X

alduxvm, 2020-09-30:
Hi all, 

I'm trying the unit .ino script for the ToF VL53L0x sensor ( https://m5stack.com/products/tof-sensor-unit ), I'm using a M5stickc unit and have them connected via the i2c cable...

https://github.com/m5stack/M5Stack/blob/master/examples/Unit/ToF_VL53L0X/ToF_VL53L0X.ino

The serial monitor does not display anything but this:

```
                                                M5Stack initializing...OK
----- START TEST ----
```

It never finishes, it never moves on? is this an issue with my ToF sensor or with my M5stickc? or that code does not work with the m5stickc?

thanks!

gnlcosta, 2020-10-04:
Hi Aldo,
I use ATOM and the tof-sensor-unit works. To use the code from ToF_VL53L0X.ino I enabled the I2C using this code:
Wire.begin(26, 32, 200000);
In your case I [suppose](https://static-cdn.m5stack.com/image/product_jpg/assets/img/product_pics/core/minicore/m5stickc/m5stickc_05.jpg):
Wire.begin(32, 33, 200000);

Ciao.

alduxvm, 2020-10-04:
Thanks Gianluca, what worked at the end was to use the example of ToF inside the M5stickC and then adding your suggestion! the code is as follows:

```
// please install vl53l0x lib first (https://github.com/pololu/vl53l0x-arduino)
// lib in Sketch->Includ Library->Library Manager, search for vl53l0x, Author: pololu

#include "M5StickC.h"
#include <Wire.h>
#include <VL53L0X.h>

VL53L0X sensor;
TFT_eSprite img = TFT_eSprite(&M5.Lcd); 

void setup() {
  Serial.begin(115200);
  Wire.begin(32, 33, 200000);
  
  M5.begin();

  img.createSprite(160, 80);
  img.fillSprite(BLACK);
  img.setTextColor(WHITE);
  img.setTextSize(2);

  sensor.setTimeout(500);
  if (!sensor.init()) {
    img.setCursor(10, 10);
    img.print("Failed");
    img.pushSprite(0, 0);
    Serial.println("Failed to detect and initialize sensor!");
    while (1) {}
  }
  // Start continuous back-to-back mode (take readings as
  // fast as possible).  To use continuous timed mode
  // instead, provide a desired inter-measurement period in
  // ms (e.g. sensor.startContinuous(100)).
  sensor.startContinuous();
}

void loop() {
  uint16_t distance = sensor.readRangeContinuousMillimeters();
  Serial.print(distance);
  if (sensor.timeoutOccurred()) { Serial.print(" TIMEOUT"); }
  Serial.println();
  img.fillSprite(BLACK);
  img.setCursor(10, 10);
  img.print(distance);
  img.pushSprite(0, 0);
}
```

The original example comes with `Wire.begin(0, 26, 100000);` and it does not work... changing to `Wire.begin(32, 33, 200000);` made it work! Thanks!

Gitshaoxiang, 2020-10-27:
M5StickC with the `ToF Unit`. you could use this one:
https://github.com/m5stack/M5StickC/tree/master/examples/Unit/TOF_VL53L0X

closed 2020-10-27 by Gitshaoxiang