m5stack/M5Stack #235: Changing I2C pins 21,22 to another pin

Chris-afking, 2020-11-06:
Hi, I am using an ESP32 and MPU6050 for gesture recognition using the Magic Wand code. All this while I have used the default I2C pins 21 and 22, but now I want to use pins 34 and 33. I have tried changing the pin declarations in the M5Stack libraries to "wire1.read (34,33)" but I only receive a single line of readings from the MPU6050, so the gesture recognition cannot work. Could you kindly teach me how to change the I2C pins so that it will work? Thank you!

felmue, 2020-11-06:
Hello @Chris-afking 

in the M5Stack device GPIO33 is internally wired to the reset line of the LCD so using it for I2C most likely won't work.

that said, in general setting GPIOs for I2C is done by calling `Wire.begin(SDA, SCL);`. In the M5Stack library this call happens inside the `M5.begin()` function.

Cheers
Felix

Chris-afking, 2020-11-09:
Alrights, I will try it out thanks! @felmue

closed 2020-11-09 by Chris-afking