m5stack/M5Stack #237: M5 ATOM Wifi power issue with RS485 to TTL Converter

peperoca116, 2020-12-10:
Hi everyone,

I'm having a very weird problem using the RS485 to TTL Converter and the M5 atom.
Whenever I plug the device with the usb it immediately connects to my wifi. 
But, when I power the device with the RS485 converter bought from the same M5stack webpage (12 v applied, neither power issues nor peaks tested with oscilloscope) the device never connects to wifi. I have tested the RSSI and it is ok (-65).

Then I tried three different Atom modules and three different RS485 converters and the same issue is present.
I believe there is a control pin or something extra in the usb power.

Can anyone give me some hints to solve this problem?
Or does anyone has the full schematic of the M5 ATOM in order to check the pinout and try to solve my problem?

Program (Just works with usb power, problems powering via RS485):
#include "WiFi.h"
 
String ssid = "COM-818";
String password =  "";
 
void setup() {
  Serial.begin(115200);  
  while (WiFi.status() != WL_CONNECTED) {

    WiFi.begin(ssid.c_str(), password.c_str());
    Serial.println("Connecting to WiFi..");
    delay(2000);
  }
 
  Serial.println("Connected to the WiFi network");
} 
void loop() { }

felmue, 2020-12-11:
Hello @peperoca116 

I don't own the RS485 so I tried to power the M5AtomMatrix with an external 5 volt power supply and it connected to WiFi just fine. I suggest you try the same to rule out there is an issue with RS485 converter and the 5 volts it generates.

Note: Please be careful when manually feeding 5 volts to the M5Atom as a mistake could easily destroy it.

BTW: I'd call `WiFi.begin()` outside and before the loop. Here is my test code (includes LED status)
```
#include <M5Atom.h>
#include "WiFi.h"

String ssid = "SSID";
String password = "PASSWORD";

void setup() {
  M5.begin(true, false, true);

  delay(100);
  M5.dis.fillpix(0x00FF00); // Red

  WiFi.begin(ssid.c_str(), password.c_str());

  while(WiFi.status() != WL_CONNECTED) {
    Serial.println("Connecting to WiFi..");
    delay(2000);
  }
  Serial.println("Connected to the WiFi network");
  M5.dis.fillpix(0xFF0000); // Green
}

void loop() {
}
```

Thanks
Felix

P.S. Shouldn't this issue be reported in M5Atom repository?

EeeeBin, 2020-12-11:
Hi peperoca116

If need  power the M5AtomMatrix with an external 5 volt,

please do not exceed 5v  (such as 5.1v), otherwise it will cause the degradation of wifi performance

hope it can help you

felmue, 2020-12-13:
Hello @EeeeBin 

I'd like to understand your comment about not exceeding 5 volts and how +0.1 volts could affect wifi performance.

I'd expect the 5 volts rail being regulated down to 3.3 volts (to power the ESP32) irrespective of whether the input voltage actually is 5 volts or 5.1 volts.

What am I missing?

Thanks
Felix

peperoca116, 2020-12-14:
So apparently it was a source power issue, some M5 modules needed more "power punch". I did a lot of testing on weekend and connected the devices with different configurations. On one hand I discovered the M5 ATOM With matrix leds only worked if you use the tail that has two power inputs and connected successfully to the network every time. On the other hand I discovered that the M5 basic atom can operate properly and connects with the modbus tail that has only one power supply/communication connector. I'll leave the post here in case anyone has the same problem.

segdy, 2024-01-09:
Did you ever figure out more? I have exactly the same issue. It's so frustrating! I even purchased a new RS485 Base adapter because I thought it was broken.

I am using the [Atom RS485 Kit](https://docs.m5stack.com/en/atom/atomic485).

I measure exactly 13V at the RS485 orange adapter (between "GND" and "DC12V"). I measure exactly 4.84V at the 5V pin (unloaded, without the Atom).

With the Atom installed I can't measure the 5V pin but I can still measure the DC12V pin and it stays at 13V. If I were drawing too much current for instance, that voltage would drop. But it doesn't, so I think the supply should be ok.