m5stack/M5Stack #241: How about to separate M5Display as a library

asukiaaa, 2021-02-13:
Thank you for sharing useful library.
M5Display saves my time a lot.

I want to use the library for other lcd but it seems difficult. ( `TFT_WIDTH` and `TFT_HEIGHT` are embedded in code.)
How about to separate M5Display as a independent library and make it one of dependent library of this lib?
If it is separated, M5Core2 and M5StickC can depend on the library and we can redule amount of code.

Arduino framework accepts `depends` option from Arduino IDE 1.8.10
https://github.com/arduino/Arduino/pull/8600
https://arduino.github.io/arduino-cli/library-specification/

My idea of definitions.

M5Display
```cpp
M5Display
{
  public:
    M5Display(uint16_t width, uint16_t height, int pinLed = -1, uint16_t ledPWMChannel = 7);
};
```

library.properties of M5Stack
```
depends=M5Display
```

M5Stack
```cpp
class M5Stack
{
  public:
    M5Stack();

    // LCD
    M5Display Lcd = M5Display(TFT_WIDTH, TFT_HEIGHT, TFT_BG, BLK_PWM_CHANNEL);
}; 

```

EeeeBin, 2021-02-13:
Hi, asukiaaa
m5stack tft is based on bobmer's tft library -> https://github.com/Bodmer/TFT_eSPI
This is easier to port to other lcd
Hope this helps you

closed 2021-02-13 by EeeeBin

asukiaaa, 2021-02-13:
Thank you for replying.
I know that library but M5Display has its [own functions](https://github.com/m5stack/M5Stack/blob/master/src/M5Display.cpp) so I think is is useful if it is separated as a library.

lovyan03, 2021-02-13:
hello @asukiaaa 
If you want, you can try this library.

https://github.com/lovyan03/LovyanGFX

Most of the functions of M5Display can be used without modification, and it works faster.

asukiaaa, 2021-02-13:
@lovyan03 
Thank you for the information.
I try it.