m5stack/M5Stack #246: Mixed licenses within M5 library components

nmori, 2021-04-13:
The M5 library is declared to be under the MIT license, but if you look at the modules, you will see that there are several licenses mixed in. (For example, GPL and CC)

Since the library is not distributed as binary code, we believe that even if it is declared as MIT, it actually needs to be treated as GPL.

If this is not what is intended, it would be better to organize and unify the licenses.

ikedam, 2021-05-02:
Same issue in M5EPD (m5paper): m5stack/M5EPD#21

nmori, 2021-07-17:
Power class, but at first I added this header because other files were registered as LGPL, and I recognized that the overall license was LGPL.

However, since the project is called MIT, I would like to make this program MIT if other modifiers allow it.
(If this is allowed, that is.)

Also, I would like to rewrite the M5.Comm class because of its small size, and PR it as an MIT licensed file.