m5stack/M5Stack #257: Unable to build Factory Test.

g0lgs, 2021-08-15:
If I try to build the Factory Test code (Arduino IDE 1.8.15 on Pi4B 32bit or Windows 10 64bit ) I get:

sketch/FactoryTest.ino.cpp.o:(.literal._Z11startupLogov+0x0): undefined reference to `m5stack_startup_music'

sketch/FactoryTest.ino.cpp.o:(.literal._Z11startupLogov+0x8): undefined reference to `gImage_logoM5'

At least some of the other examples build fine.

Tinyu-Zhao, 2021-08-16:
Try putting both files in the same path as factorytest.ino
[bmp_map.c](https://github.com/Tinyu-Zhao/M5Stack/blob/master/examples/Basics/FactoryTest/bmp_map.c)
[startup_music.c](https://github.com/Tinyu-Zhao/M5Stack/blob/master/examples/Basics/FactoryTest/startup_music.c)
like this .
[image]

g0lgs, 2021-08-16:
That sorted it - Thanks.

closed 2021-08-16 by g0lgs