m5stack/M5Stack #259: M5Stack@: dependency 'M5_BM8563' is not available

sascha1337, 2021-08-24:
```
Name: "M5Stack"
  Author: M5Stack
  Maintainer: M5Stack
  Sentence: Library for M5Stack Core development kit
  Paragraph: See more on http://M5Stack.com
  Website: https://github.com/m5stack/m5stack
  Category: Device Control
  Architecture: esp32
  Types: Contributed
  Versions: [0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.5, 0.1.6, 0.1.8, 0.1.9, 0.2.0, 0.2.1, 0.2.2, 0.2.4, 0.2.5, 0.2.6, 0.2.7, 0.2.8, 0.2.9, 0.3.0, 0.3.1, 0.3.2, 0.3.3, 0.3.4, 0.3.5, 0.3.6]
  Provides includes: M5Stack.h
  Dependencies: M5GFX, UNIT_ENV, UNIT_4RELAY, ADXL345, FastLED, MODULE_GRBL13.2, Adafruit MCP4725, Adafruit TCS34725, Adafruit NeoPixel, MAX30100lib, MFRC522_I2C, M5_BM8563, M5_ADS1100, M5_ADS1115, HX711 Arduino Library, PCA9554
  ```

**M5_BM8563** <-- My arduino-cli is not able to resolve that dependency.

```
❯ arduino-cli lib list
Name                 Installed Available Location              Description                       
Adafruit_GFX_Library 1.10.10   -         LIBRARY_LOCATION_USER -                                 
...
M5_BM8563            0.0.1     -         LIBRARY_LOCATION_USER -       
```

Manually installed from git, recognized in arduino-cli, still not able to install the m5stack library due to the same error:

```
❯ arduino-cli lib install M5Stack
Error installing M5Stack: Error resolving dependencies for M5Stack@: dependency 'M5_BM8563' is not available
```

sascha1337, 2021-11-04:
still not updated? wuh guys cleaning up my git, seeing this thing still not fixed

MatteoGheza, 2021-11-04:
Fixed using the release archive as a source.
For example: https://github.com/m5stack/M5StickC/archive/refs/tags/0.2.4.zip
If you want to see a working arduino compile-sketches GitHub Actions, see https://github.com/josephdadams/TallyArbiter/blob/move_listener_clients/.github/workflows/build-listener-clients.yaml#L18

sascha1337, 2021-11-06:
Thanks mate

Tinyu-Zhao, 2022-07-04:
I just tested and this repository works fine being downloaded.
[image]

closed 2022-07-04 by Tinyu-Zhao