m5stack/M5Stack #295: the system cannot find the file specified

skywo1f, 2022-10-29:
### Describe the bug

on compilation in arduino (any m5stack example), I get the following error:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\iviti\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\iviti\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\iviti\Documents\Arduino\libraries -fqbn=m5stack:esp32:m5stack-core2:PSRAM=enabled,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=all -vid-pid=1A86_55D4 -ide-version=10813 -build-path C:\Users\iviti\AppData\Local\Temp\arduino_build_592585 -warnings=none -build-cache C:\Users\iviti\AppData\Local\Temp\arduino_cache_675155 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\esptool_py\3.3.0 -prefs=runtime.tools.esptool_py-3.3.0.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\esptool_py\3.3.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mklittlefs.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -verbose C:\Users\iviti\Documents\Arduino\libraries\M5Stack-master\examples\Games\Tetris\Tetris.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\iviti\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\iviti\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\iviti\Documents\Arduino\libraries -fqbn=m5stack:esp32:m5stack-core2:PSRAM=enabled,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=all -vid-pid=1A86_55D4 -ide-version=10813 -build-path C:\Users\iviti\AppData\Local\Temp\arduino_build_592585 -warnings=none -build-cache C:\Users\iviti\AppData\Local\Temp\arduino_cache_675155 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\esptool_py\3.3.0 -prefs=runtime.tools.esptool_py-3.3.0.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\esptool_py\3.3.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mklittlefs.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\tools\xtensa-esp32s3-elf-gcc\gcc8_4_0-esp-2021r2-patch3 -verbose C:\Users\iviti\Documents\Arduino\libraries\M5Stack-master\examples\Games\Tetris\Tetris.ino
Using board 'm5stack-core2' from platform in folder: C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.5+1.0
Using core 'esp32' from platform in folder: C:\Users\iviti\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.5+1.0
cmd /c if exist "C:\\Users\\iviti\\Documents\\Arduino\\libraries\\M5Stack-master\\examples\\Games\\Tetris\\partitions.csv" COPY /y "C:\\Users\\iviti\\Documents\\Arduino\\libraries\\M5Stack-master\\examples\\Games\\Tetris\\partitions.csv" "C:\\Users\\iviti\\AppData\\Local\\Temp\\arduino_build_592585\\partitions.csv"
cmd /c if not exist "C:\\Users\\iviti\\AppData\\Local\\Temp\\arduino_build_592585\\partitions.csv" if exist "C:\\Users\\iviti\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.5+1.0\\variants\\m5stack_core2\\partitions.csv" COPY "C:\\Users\\iviti\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.5+1.0\\variants\\m5stack_core2\\partitions.csv" "C:\\Users\\iviti\\AppData\\Local\\Temp\\arduino_build_592585\\partitions.csv"
cmd /c if not exist "C:\\Users\\iviti\\AppData\\Local\\Temp\\arduino_build_592585\\partitions.csv" COPY "C:\\Users\\iviti\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.5+1.0\\tools\\partitions\\default_16MB.csv" "C:\\Users\\iviti\\AppData\\Local\\Temp\\arduino_build_592585\\partitions.csv"
The system cannot find the file specified.
exit status 1
Error compiling for board M5Stack-Core2.

### To reproduce

download the board and example libraries, try to compile

### Expected behavior

successful compilation of basic examples

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Knight13th, 2022-10-29:
Also seeing the same error.

skywo1f, 2022-10-30:
I updated to the newest arduino, and updated all of the libraries. Still getting the same error.

Tinyu-Zhao, 2022-10-30:
Did you choose the right development version, because I see in the compilation log you provided
[image]

This is the same example I compiled as yours, and it works fine.
[image]

skywo1f, 2022-10-30:
are you saying that the m5stack library is not compatible with the m5stack-core2?

Tinyu-Zhao, 2022-10-30:
> are you saying that the m5stack library is not compatible with the m5stack-core2?

M5stack compilation should be done by M5Stack-Core-ESP32

skywo1f, 2022-10-30:
I changed the compilation board setting to M5Stack-Core-ESP32, but I get the same error

HB9DTX, 2022-11-04:
Give it a try with an older version of M5stack board Manager. I had a similar issue using an old board. Version 1.0.7 of M5stack worked while 2.0.5+1.0 didn't work.

Tinyu-Zhao, 2022-11-06:
You can try the latest version 2.0.5_1.0

closed 2023-02-05 by Tinyu-Zhao