m5stack/M5Stack #296: any valid example for M5Stack Core

ExtremeGTX, 2022-11-07:
### Describe the bug

I tried to use the following code to play some wavfiles but unfortunatly it didn't work.

The MicroPython runtime keep complaining about ` I2S.MODE_MASTER | I2S.MODE_TX | I2S.MODE_DAC_BUILT_IN` that neither of them is part of `I2S` module. could you please point me to an example which works on [m5stack Core basic (The EOL version)](https://shop.m5stack.com/products/basic-core-iot-development-kit).

```
from machine import I2S, SDCard, Pin
from m5stack import *
import os, uos
from wav import wave

#mount the sd card
sd = SDCard(slot = 2, sck = Pin(23), miso = Pin(33), mosi = Pin(19), freq = 10000000)

#initialize the I2S device
i2s = I2S(  mode = I2S.MODE_MASTER | I2S.MODE_TX | I2S.MODE_DAC_BUILT_IN,
            rate = 16000,
            bits = 16,
            channel_format = I2S.CHANNEL_ONLY_LEFT,
            data_format = I2S.FORMAT_I2S_MSB)

#create a function to play the wav
def wav_player(fname):
    wav = wave.open(fname)
    i2s.set_dac_mode(I2S.DAC_RIGHT_EN)
    i2s.sample_rate(wav.getframerate())
    i2s.bits(wav.getsampwidth() * 8)
    i2s.nchannels(wav.getnchannels())
    i2s.volume(20)

    while True:
        data = wav.readframes(1024)
        if len(data) > 0:
            i2s.write(data)
        else:
            wav.close()
            break

# Playing WAV audio file
lcd.clear()
lcd.print('working',0,0,0xffffff)

try:
    uos.mountsd(sd, '/sd')
except:
    #os.mountsd()
    pass

speaker.setVolume(2)
#kill the weird start noise glitch
i2s.stop()

#point the directory to the files you have on your sd card - works best with short low file size wavs
while True:
    if btnA.wasPressed():
        wav_player('/flash/res/yourfile.wav')
        i2s.stop()
    if btnB.wasPressed():
        wav_player('/sd/yourOtherFile.wav')
        i2s.stop()
    if btnC.wasPressed():
        wav_player('/sd/anotherfile.wav')
        i2s.stop()

```

### To reproduce

1. Use M5Burner-v3-beta-win-x64 to flash UiFlow v1.10.7.1 to the m5stack core basic
2. Go to https://flow.m5stack.com/ 
3. switch to python mode
4. Copy the code above
5. Click `Run`
6. Check the device screen for the related python error

### Expected behavior

The code should run without displaying python errors.

### Environment

- OS: Windows 10
- IDE &IDE Version: [UiFlow v1.10.7](https://flow.m5stack.com/) 
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2022-11-09:
@imliubo Please help check this.

chipguyhere, 2024-01-26:
I don't think Core has any I2S hardware in the first place.  I think it plays analog audio directly from ESP32 pin 25, which supports direct digital-to-analog output.  There's a Speaker class in the library that looks like it can play wave audio just by loosely banging it out to the pin.

floriansenf, 2024-12-24:
DId someone find a solution? Facing the same problem.