m5stack/M5Stack #305: PM25 KIT Example Error

Zazucki, 2023-02-23:
### Describe the bug

When I am compiling PM25.ino I get the following error, where can I find this UNIT_ENV.h file? I think I have all of the libraries that I need installed.

```
/Users/jk/Documents/M5Stack/examples/KIT/PM25/PM25.ino:4:10: fatal error: UNIT_ENV.h: No such file or directory
 #include "UNIT_ENV.h"
          ^~~~~~~~~~~~
compilation terminated.
Multiple libraries were found for "SD.h"
  Used: /Users/jk/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.6/libraries/SD
  Not used: /Users/jk/Library/Arduino15/libraries/SD
exit status 1

Compilation error: UNIT_ENV.h: No such file or directory
```

### To reproduce

I cloned the repo and opened the PM25.ino file in Arduino IDE, then click compile.

### Expected behavior

Finds UNIT_ENV.h file

### Screenshots

[image]

### Environment

- OS: macOS Ventura 13.2.1 (22D68)

- IDE &IDE Version:
Version: 2.0.3
Date: 2022-12-05T09:29:06.685Z
CLI Version: 0.29.0 [76251df9]

Copyright © 2023 Arduino SA

- Repository Version:
clone was today.

### Additional context

I recently purchased the PM2.5 Air Quality Kit（SHT30）from m5stack.com

It came flashed with what appeared to be working code, then the temp and humidity stopped working / showed only 0 as output. So I decided to try flashing the newest code because I saw there are recent changes. This is where I run into this error.

I saw in some other posts that changing this to M5_ENV.h works, and while that does get the code to compile, the code does not work. It shows PM2.5 in a blue bar at the top, but nothing else is displayed on the screen, just black.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Zazucki, 2023-02-23:
I realize now that I didn't have the stand plugged in which contains the SHT30 and that's why they were reading 0 before. So I did not actually need to put any new code on the device, it was working. My problem now is that I do not have the original code it came with, and the example given does not work.

Tinyu-Zhao, 2023-02-26:
Thank you for your feedback, I have updated the example, please try to compile it with the latest [example](https://github.com/m5stack/M5Stack/tree/master/examples/KIT/PM25).

closed 2023-02-26 by Tinyu-Zhao