m5stack/M5Stack #309: Compatibility between M5Stack and M5Core2 sketches

reflectingme, 2023-03-20:
Maybe I have this wrong, but I would have thought that there would be compatibility between sketches in the M5Stack examples and the M5Core2 device... sadly, alas not.  

I have tried the examples/TFT_Clock sketch and the code compiles and uploads just fine on my M5Core2 device, but then the device just hangs.  

Are there any straightforward settings/configs that I would need to adjust or libraries I'd need to add in order to allow this code to work with the m5core2, or is a total re-write of the code required due to 100% non compatibility?

IcingTomato, 2023-06-15:
We recommand you to using M5Unified.
https://github.com/m5stack/M5Unified

Tinyu-Zhao, 2023-06-18:
Many of the pins between Core2 and M5Stack devices are different, and M5Stack has several hosts, so we recommend that you use M5Unified for inter-device compatibility.

[image]

closed 2023-06-18 by Tinyu-Zhao