m5stack/M5Stack #311: Arduino Library for Interacting with ATECC608B

aecorredor, 2023-05-16:
### Describe the bug

Trying to interact with the secure module in the M5Stack using the Arduino framework, but I am struggling to find an up to date library for doing so. The closest I've found is https://github.com/GabrielNotman/cryptoauthlib, but that hasn't been updated in 4 years and does not seem to support the `B` variant.

### To reproduce

Run any project on the `espressif32` platform with the `arduino` framework and try interacting with the secure hardware module (ATECC608B).

### Expected behavior

Able to read/write from the ATECC608B chip. (Maybe at least write once then only read?).

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2023-05-18:
Sorry, there is no ATECC608B chip in M5Stack, but in M5Core2. Please confirm your device type first.

aecorredor, 2023-05-18:
@Tinyu-Zhao sorry, just corrected it. I meant `M5Core2`. Thanks!

Tinyu-Zhao, 2023-05-26:
> 

You can try https://github.com/arduino-libraries/ArduinoECCX08.