m5stack/M5Stack #314: M5Stack Library won't install on Arduino IDE

JaimeCababie, 2023-09-10:
### Describe the bug

M5 Stack library 0.4.5 wont install when using the library manager because 'MODULE_GRBL13.2' is not available. 

### To reproduce

1. open library manager
2. search for M5Stack Library 
3. Try to install M5Stack v. 0.4.5 

### Expected behavior

I would expect for the library to install

### Screenshots

[image]

### Environment

Arduino IDE 2.2.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

srolf, 2023-09-10:
Same for me, any version of "M5Stack" library 0.4.1 is causing such a dependency problem.

CitrusPeel, 2023-09-12:
The package's name "MODULE_GRBL13.2" in [library.properties](https://github.com/m5stack/M5Stack/blob/master/library.properties) should be "MODULE_GRBL_13.2"

sigmaeo, 2023-10-04:
this is fixed in M5StackCore2 0.1.7

closed 2024-06-20 by Tinyu-Zhao