m5stack/M5Stack #317: Compilation errors

Richrfl, 2023-10-27:
### Describe the bug

I have had compilation issues when calling the library and I had always compilation errors. Therefore, I cleaned up my libraries, and started a new project but still I have errors that I cannot explain or solve.
Here is the project:

```cpp
# include <M5Stack.h>;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

```
and here is the error report:

In file included from c:\Users\rcric\Documents\Arduino\libraries\M5Stack\src/M5Display.h:8,
                 from c:\Users\rcric\Documents\Arduino\libraries\M5Stack\src/M5Stack.h:111,
                 from C:\Users\rcric\AppData\Local\Temp\.arduinoIDE-unsaved2023927-53880-130vt5o.cddq\sketch_oct27a\sketch_oct27a.ino:1:
c:\Users\rcric\Documents\Arduino\libraries\M5Stack\src/utility/In_eSPI.h:633:20: error: 'VSPI' was not declared in this scope
     uint8_t port = VSPI;
                    ^~~~
c:\Users\rcric\Documents\Arduino\libraries\M5Stack\src/utility/In_eSPI.h:633:20: note: suggested alternative: 'SPI'
     uint8_t port = VSPI;
                    ^~~~
                    SPI

exit status 1

Compilation error: exit status 1

I would appreciate any suggestion to fix this issue. I guess it's on my side, but I can't figure it out.

### To reproduce

Create project
Compile
Board selected Adafruit QT Py ESP32-S3 No PSRAM.

### Expected behavior

Success compilation

### Environment

- OS:
- IDE &IDE Version: 2.2.1
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

jackiehanlon, 2024-02-01:
I had the same issue at first- this is simply due to using the incorrect library for your device. ESP32-S3 devices do not support VSPI, you will want to use the M5CoreS3 library rather than the M5Stack library. See [this reddit post](https://www.reddit.com/r/esp32/comments/197zhss/issues_compiling_for_my_m5stack_cores3_vspi_error/) about the same issue for reference.

Tinyu-Zhao, 2024-06-20:
I had the same issue at first- this is simply due to using the incorrect library for your device. ESP32-S3 devices do not support VSPI, you will want to use the M5CoreS3 library rather than the M5Stack library. See [this reddit post](https://www.reddit.com/r/esp32/comments/197zhss/issues_compiling_for_my_m5stack_cores3_vspi_error/) about the same issue for reference.

Thank you for your answer.

closed 2024-06-20 by Tinyu-Zhao