m5stack/M5Stack #323: please What is the meaning of this? How can it be solved?

mostafanafie5, 2024-02-21:
### Describe the bug

please What is the meaning of this? How can it be solved?

### To reproduce

please What is the meaning of this? How can it be solved?

### Expected behavior

please What is the meaning of this? How can it be solved?

### Screenshots

[image]

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-06-21:
Please attach your program, without providing more details, it is impossible to troubleshoot.

Tinyu-Zhao, 2024-07-03:
The questioner cannot provide useful information, so it is closed.

closed 2024-07-03 by Tinyu-Zhao