m5stack/M5Stack #331: How can i connect the two units by I2C?

Mbincao, 2024-12-06:
### Describe the bug

only one can work,

### To reproduce

https://github.com/user-attachments/assets/7dda8e0b-280e-4a14-857a-5bb56a66fd16

### Expected behavior

[image]

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-12-06:
joystick2.begin (&Wire,JOYSTICK2_ADDR, 2, 1) ；//portA 2,1 //portB 9,8 //portC 17,18

change to 
joystick2.begin (&Wire1,JOYSTICK2_ADDR, 2, 1) ；//portA 2,1 //portB 9,8 //portC 17,18

Tinyu-Zhao, 2024-12-06:
Can't see your picture
Try again without using the M5CoreS3 library.

Mbincao, 2024-12-06:
I commented on these lines of code.the power of control.uart can get the value,but why the display can't
    // //配置电源
    // auto cfg = M5.config();
    // // cfg.output_power = true;
    // CoreS3.begin(cfg);
    // CoreS3.Display.setTextSize(2);

Tinyu-Zhao, 2024-12-06:
😂Is there a possibility that M5CoreS3 can control the screen? Of course it cannot be used as you commented.

Tinyu-Zhao, 2024-12-06:
You are using I2C communication, why did UART suddenly appear?

Mbincao, 2024-12-06:
oh,sorry,i use the Serial Port Debugging Assistant to see the result

Tinyu-Zhao, 2024-12-06:
M5CoreS3 internally calls M5Unified, M5, and a series of initialization operations are performed when begin. A certain set of I2C may be enabled by default, causing conflicts. Therefore, you cannot use Wire and Wire1 at the same time after enabling M5.begin.

Mbincao, 2024-12-06:
i mean  when i didn't comment the code,it's value didn't change as i control the joysticks

Tinyu-Zhao, 2024-12-06:
也许你可以尝试
sensor.begin (&Wire1,SCROLL_ADDR, 17, 18) 
joystick2.begin (&wire,JOYSTICK2_ADDR, 2,1)

Mbincao, 2024-12-06:
when you change another,the other doesn't work,it's so wired

Tinyu-Zhao, 2024-12-06:
那是不是如果不使用M5CoreS3这个库,两个模块都可以用

Mbincao, 2024-12-07:
可能是调用了这个显示屏的缘故，我注释了显示屏供电后串口就可以正确读取数据了

Tinyu-Zhao, 2024-12-10:
显示屏用的是M5Unified, M5.Begin后 Wire1默认被使用, 所以你再用Wire1可能就会有问题了

Mbincao, 2024-12-17:
主要想问问是传输数据有问题吗，具体该咋解决？