m5stack/M5Stack #335: PbHub only displays one RGB led despite setting more than one on a channel

dpharris, 2025-05-14:
### Describe the bug

PbHub is supposed to display on a chain of RGB smart LEDs per channel.  However, despite setting the number of LEDs, only the first LED lights, albeit with the last value in the chain.  

David

### To reproduce

..

### Expected behavior

..

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Stack/issues?q=)
- [x] My report contains all necessary details

dpharris, 2025-05-14:
This is as per the comment on the product page: 
**I/O Hub 1 to 6 Expansion Unit (STM32F0)**
Works almost as advertised. As I understand the description in RGB mode the max. number or RGB LEDs is 74. However only the first RGB LED can be addressed and used.

Has this been addressed?  

David

Tinyu-Zhao, 2025-05-16:
Are you using this example? Have you modified the code? Which host model are you using? What model of lamp beads are you using? Without detailed information, it is difficult for me to locate the problem.

dpharris, 2025-05-22:
Hi Tinyu-Zhao --

Thank you for your reply.

I am using:
a M5Stack Atom
a PbHub v1.1 STM32F030
and the example *rgb_led.ino* included with the library at:
https://github.com/m5stack/M5Unit-PbHub
modified for the Atom, and using a 10-LED chain, see below.

Problem: It does not fill all the LEDs, but only the first one.
This problem had been previously pointed out in the comments for this unit
at https://shop.m5stack.com/products/i-o-hub-1-to-6-expansion-unit-stm32f0
[image: image.png]
Thanks for your attention to this matter,
David

/**
* @file rgb_led.ino
* @author SeanKwok ***@***.***)
* @brief
* @version 0.1
* @date 2024-03-05
*
*
* @Hardwares: M5Core + PbHub / PbHub v1.1
* @Platform Version: Arduino M5Stack Board Manager v2.1.0
* @Dependent Library:
* M5UnitPbHub: https://github.com/m5stack/M5Unit-PbHub
*/

#include "M5UnitPbHub.h"

M5UnitPbHub pbhub;

void setup() {
Serial.begin(115200);
//if (!pbhub.begin(&Wire, UNIT_PBHUB_I2C_ADDR, 21, 22, 100000U)) {
if (!pbhub.begin(&Wire, UNIT_PBHUB_I2C_ADDR, *26, 32*, 100000U)) {
Serial.println("Couldn't find Pbhub");
while (1) delay(1);
}
for (uint8_t ch = 0; ch < 6; ch++) {
pbhub.setLEDNum(ch, 74);
pbhub.setLEDBrightness(ch, 40);
}
}
void loop() {
// ch: 0-5
// index: 0-1
// status: 0/1
for (uint8_t ch = 0; ch < 6; ch++) {
pbhub.fillLEDColor(ch, 0, 73, 0xff0000);
}
delay(1000);
for (uint8_t ch = 0; ch < 6; ch++) {
pbhub.fillLEDColor(ch, 0, 73, 0x00ff00);
}
delay(1000);
for (uint8_t ch = 0; ch < 6; ch++) {
pbhub.fillLEDColor(ch, 0, 73, 0x0000ff);
}
delay(1000);

for (uint8_t ch = 0; ch < 6; ch++) {
for (uint8_t index = 0; index < 73; index++) {
pbhub.setLEDColor(ch, index, 0x00ff00);
delay(20);
}
}
delay(1000);
}

Tinyu-Zhao, 2025-05-23:
I reproduced the problem using SK6812. If your lamp is also SK6812, just follow my instructions.

1. Use the latest [example](https://github.com/m5stack/M5Unit-PbHub/blob/main/examples/rgb_led/rgb_led.ino) first
2. If the firmware version is less than 2, it does not support SK6812 lamp beads, and you need to upgrade the Unit PbHub firmware to V2
3. For detailed upgrade tutorials, please visit this [website](https://docs.m5stack.com/en/guide/develop_tools/daplink).